JeffersonLab/coatjava #33: Some small issues in DC reconstruction

tongtongcao, 2023-05-17:
There are small issues in codes for DC reconstruction. 
1. Comment are not consistent with codes in [https://github.com/JeffersonLab/clas12-offline-software/blob/development/reconstruction/dc/src/main/java/org/jlab/rec/dc/cluster/ClusterCleanerUtilities.java#L760](https://github.com/JeffersonLab/clas12-offline-software/blob/development/reconstruction/dc/src/main/java/org/jlab/rec/dc/cluster/ClusterCleanerUtilities.java#L760)
2. Since _OutOfTimeFlag is always false by default for hit-based tracking, Out-of-time remover did nothing in [https://github.com/JeffersonLab/clas12-offline-software/blob/development/reconstruction/dc/src/main/java/org/jlab/rec/dc/cluster/ClusterFinder.java#L207](https://github.com/JeffersonLab/clas12-offline-software/blob/development/reconstruction/dc/src/main/java/org/jlab/rec/dc/cluster/ClusterFinder.java#L207)
3. Unused variable is defined in [https://github.com/JeffersonLab/clas12-offline-software/blob/development/reconstruction/dc/src/main/java/org/jlab/rec/dc/hit/FittedHit.java#L669-L671](https://github.com/JeffersonLab/clas12-offline-software/blob/development/reconstruction/dc/src/main/java/org/jlab/rec/dc/hit/FittedHit.java#L669-L671)

Fixing these issue will not affect results at all.
See comparison for hit-based particles and time-based particles before and after issue fixing.
[diff of RECHBParticle.pdf](https://github.com/JeffersonLab/clas12-offline-software/files/10439971/diff.of.RECHBParticle.pdf)
[diff of RECParticle.pdf](https://github.com/JeffersonLab/clas12-offline-software/files/10439974/diff.of.RECParticle.pdf)

baltzell, 2025-04-10:
this can probably be closed by now?

tongtongcao, 2025-04-10:
Yes. The issue can be closed.

closed 2025-04-10 by tongtongcao