JeffersonLab/coatjava #173: Fix helicity decoder.

mholtrop, 2024-01-21:
The CodaEventDecoder::getDataEntries_HelicityDecoder only works correctly when the EVIO data is corrupted and puts 40 helicity decoder board banks once every 40 events.
In the correct DAQ configuration there should be a single 57651 Helicity Decoder Board bank in _**every**_ event. In this case the EVIO bank does not have the 'BLKHDR' word, so decoding is incorrect and causes a crash.

Run 019400 and 019401 had the helicity data in every event. The first event from file clas_019400.evio.00040 decoded with evio2xml has for bank 57651 (= 0xe133):

         <uint32 data_type="0x1" tag="57651" padding="0" num="39" length="19" ndata="18">
            0x94f4de10  0x9d2815ef  0x000004f5  0xc000000e  
            0x0ec76249  0x0004b336  0x0004b337  0x00012cce  
            0x0002599c  0x0027d519  0x0026e0fa  0x0000f41e  
            0x003f926a  0x00000130  0xeeeeeeee  0xaaaaaaaa  
            0xd2cd3332  0x8ec76249  
         </uint32>

First word is "EVTHDR", second and third are "TRGTIME", forth is "DECODER DATA", indicating the next 14 (0x0e) words are the readout of the Helicity Decoder.

This issue is related to issue #171

baltzell, 2024-01-23:
Running this on one of the few files I've been using for testing, looks to me like the bit sequences are now off in the HEL::decoder bank.  For example:

```
Pair:      11011101110111011101110111011101
Pattern:   00000000000000000000000000000110
Helicity:  01010101010101010101010101010101
HelicityP: 10011010010110011001101001011010
```

Previously they were all consitent with good quartet patterns, assuming signal inversion (which has now been confirmed).

baltzell, 2024-01-30:
closed by #174

closed 2024-01-30 by baltzell