JeffersonLab/coatjava #222: hipo-utils command line arguments

whit2333, 2024-03-14:
Missing help argument.  
```
hipo-utils -h

    Usage : hipo4utils [commands] [options]

    commands:
             -info :  show the information about the file
            -merge :  Merge several files together
           -filter :  filter banks from the input files
             -hide : reduce constent of the file by applying expressions
            -stats :  prints statistics for the file
             -dump :  dump the content of the file on the screen
           -reduce : reduce constent of the file by applying expressions
         -generate :  generated command for many files
           -update : update dictionary in the file
            -split :  splits file into smaller chunks
             -test :  reads the file to check for file integrity

Choose wisely.....
*** ERROR *** : unknown command : -h

```

baltzell, 2024-03-19:
The source for that isn't in coatjava.  It's probably here:  https://github.com/jeffersonlab/jnp

whit2333, 2024-03-19:
> Closed  #222  as completed.

That is a private repo. I cannot see it even when signed in.

c-dilks, 2024-03-20:
@gavalian can you fix this in `jnp`? Also:
- `org.jlab.jnp.hipo4.utils.HipoUtilities` should exit non-zero when it fails
- error messages should stream to `stderr`, not `stdout`
- `hipo4utils` in the usage guide should be the name of the wrapper script `hipo-utils` (if you don't want this in `jnp`, we can pipe through `sed` here in `hipo-utils`)

whit2333, 2024-03-22:
What _is_ jnp?