JeffersonLab/coatjava #273: ALERT is missing DetectorTypes

baltzell, 2024-05-22:


baltzell, 2024-05-22:
and DetectorId values are inconstent with available DetectorType values, are they supposed to be the same?

whit2333, 2024-10-22:
@baltzell Do we just make up a number or what?

whit2333, 2024-10-23:
What is the relationship between the numbers in DetectorType and the "group"s found [here](https://github.com/JeffersonLab/coatjava/blob/90f57f19a8e89a24da1b0da5a7ccf9b97ea360b7/etc/bankdefs/hipo4/data.json#L492) or [here](https://github.com/JeffersonLab/coatjava/blob/90f57f19a8e89a24da1b0da5a7ccf9b97ea360b7/etc/bankdefs/hipo/DATA.json#L383)?

baltzell, 2024-11-01:
No relationship there.  Each must be unique, I think that's the only critical requirement, and the rest is just convention probably worth maintaining.

Closed by #360 and/or #361.

closed 2024-11-01 by baltzell