JeffersonLab/coatjava #293: freehep maven dependency issues

baltzell, 2024-07-11:
Started probably yesterday ...

[ERROR] Failed to execute goal on project clas12rec: Could not collect dependencies for project org.jlab.clas:clas12rec:pom:10.1.2-SNAPSHOT
[ERROR] Failed to read artifact descriptor for org.freehep:freehep-jminuit:jar:1.0.2
[ERROR] 	Caused by: The following artifacts could not be resolved: org.freehep:freehep-jminuit:pom:1.0.2 (absent): Could not transfer artifact org.freehep:freehep-jminuit:pom:1.0.2 from/to freehep-repo-public (https://srs.slac.stanford.edu/nexus/content/groups/freehep-maven2-public/): Connection reset

gavalian, 2024-07-11:
I'm getting the same thing. I wonder if this is a temporary problem? 
If not, we can install the Jminuit locally on clas12maven and use it from there.

c-dilks, 2024-07-11:
I can't access https://srs.slac.stanford.edu/nexus/content/groups/freehep-maven2-public/ and all the recent wayback machine snapshots get stuck in a loop of errors. Maybe we use a different freehep maven repo mirror?

gavalian, 2024-07-11:
If you can find one, yes. all the documentation on this is outdated. I'm having hard time finding the right one.

c-dilks, 2024-07-11:
I tried using the one from mvnrepository: https://mvnrepository.com/artifact/org.freehep/freehep-jminuit

I also removed everything that has to do with srs.slac.stanford.edu, but _something_ is still dependent on it... I'll open a draft PR so we can take a closer look at the logs.

c-dilks, 2024-07-11:
We decided to wait til Monday to see if `srs.slac.stanford` comes back; until then, nightly `clas12-validation` failures will spam our inboxes...

baltzell, 2024-07-12:
I installed all the necessary freehep maven repos on clasweb ...

PR #296 (2024-07-12): copy all freehep maven repos to clasweb and use them
closes #293
closed 2024-07-12 by c-dilks