JeffersonLab/coatjava #382: Single event tests need updating

baltzell, 2024-12-06:
replacing [this class](https://github.com/JeffersonLab/coatjava/blob/development/common-tools/clas-analysis/src/main/java/org/jlab/analysis/physics/TestEvent.java) with a HIPO file

c-dilks, 2024-12-06:
Can we have the CI generate such a HIPO file and either [cache it](https://github.com/actions/cache) or save it as an artifact? Or could we put it on xrootd (CI example [here](https://github.com/JeffersonLab/iguana/blob/92e45c6f4cd5f660a91152aea81a5314c62feff1/.github/workflows/ci.yml#L50-L78))?

baltzell, 2024-12-07:
I think that's a bigger project, for scale.  Here is just a replacement of the existing source-code event or two with a HIPO file resource.

baltzell, 2025-04-10:
done, /etc/data/test

closed 2025-04-10 by baltzell