JeffersonLab/coatjava #413: CI throughput

baltzell, 2025-01-03:
Seems getting too slow sometimes, queuing.  Is this repo a major contributor to that?  If so, we should cut back job counts significantly.

baltzell, 2025-01-03:
Probably the "validation" triggers need reduction:  https://github.com/orgs/JeffersonLab/actions/metrics/usage

c-dilks, 2025-01-05:
> Probably the "validation" triggers need reduction: https://github.com/orgs/JeffersonLab/actions/metrics/usage

Yeah, by default `clas12-validation` runs 7 event generators x 6 configurations = 42 jobs per workflow:

```json
{
  "evgen": [
    "e_K",
    "e_KC",
    "e_n",
    "e_g",
    "e_nC",
    "eFT_K",
    "e_gFT"
  ],
  "config": [
    "clas12-default",
    "rga_spring2018",
    "rga_fall2018",
    "rgk_fall2018_FTOn",
    "rgb_fall2019",
    "rgc_summer2022"
  ]
}
```

We probably don't need to run this many every time, and it indeed clogs the queue. I'm guilty of cancelling some validation jobs since they impede other CI work.

This matrix is configurable from the caller workflow ([see the README file](https://github.com/JeffersonLab/clas12-validation/blob/main/README.md)), and/or we can change the default matrix.

closed 2025-01-06 by c-dilks