JeffersonLab/coatjava #475: TI timestamp mismatch tolerance

baltzell, 2025-02-17:
Sergey suggests setting it to 5 tics.  Currently it's 1(?).

whit2333, 2025-02-18:
What is the value of the [`RUN::config` timestamp](https://github.com/JeffersonLab/coatjava/blob/development/reconstruction/tof/src/main/java/org/jlab/service/ctof/CTOFEngine.java#L100) expected to be?

baltzell, 2025-02-25:
Partially closed by #482, but later realized we should add an additional requirement that the `i0` reference (which is what goes in `RUN::config.timestamp` if "all" ROCs have "equal" TI timestamps) is not from ATOF (same as is done currently for the `tiMaster`, which is defined as the one that has the "HEAD" bank [0xe10f](https://github.com/JeffersonLab/clas12-parms/blob/b5851869891a4652ce51363af1e1b4fa783f4e8b/clonbanks.xml#L257-L268)/[57615](https://github.com/JeffersonLab/coatjava/blob/20f3d25b09e1c5f40b2f42a3aa8696040144595d/common-tools/clas-detector/src/main/java/org/jlab/detector/decode/CodaEventDecoder.java#L237-L252)), then #482 would work as intended for all data sets until we choose a better approach.

closed 2025-02-27 by whit2333