JeffersonLab/coatjava #489: recon-util returning wrong error code

whit2333, 2025-02-27:
`recon-util` doesn't return an error code when the output hipo file already exists. It returns true (0) but should return a non-zero error code.
Here is an example. https://code.jlab.org/hallb/alert/c12/-/jobs/8318#L237
This job should fail.

baltzell, 2025-02-27:
Looks like that behavior is here:  https://github.com/JeffersonLab/jnp/blob/870693eae84d1c07aedd6272c62c8b97b63ea62a/jnp-hipo4/src/main/java/org/jlab/jnp/hipo4/base/Writer.java#L251-L264

whit2333, 2025-02-27:
Where? I do not have permissions to see that proprietary software the entire CLAS collaboration relies on.

[image]

baltzell, 2025-08-21:
created & linked new issue at https://code.jlab.org/hallb/clas12/coatjava/jnp/-/issues/15