JeffersonLab/coatjava #493: AHDC and ATOF bank and variable names should be consistent

Sangbaek, 2025-02-27:
## Troubles

In [alert.json](https://github.com/JeffersonLab/coatjava/blob/development/etc/bankdefs/hipo4/alert.json), the bank and variable names would better to follow the consistent capitalization.

- AHDC::Clusters and ATOF::clusters are similar banks, but differ in capitalization
- AHDC::Hits and ATOF::hits are similar banks, but differ in capitalization
- AHDC::Hits:ID and ATOF::hits:id are similar variables but differ in capitalization
- AHDC::Hits:ID and AHDC::Hits:Doca start with upper cases, but AHDC::Hits:layer, AHDC::Hits:superlayer, AHDC::Hits:wire start with lower cases.

## Why this is a problem?

The inconsistency in naming rules can induce errors in code, especially when collaborators besides who named these banks need to develop some codes, or analyze the data.

## Suggestions

All bank names should follow the rule: 
[Uppercase]:[Lowercase]:[Lowercase], i.e., e.g.) AHDC::hits:x, to be consitent with most of bankdefs.

baltzell, 2025-04-09:
If this is going to happen, it needs to be done before requesting access to these banks in CED.

Sangbaek, 2025-04-11:
@ftouchte , just checking, is this still on your to-do list?

ftouchte, 2025-04-11:
Yes, it is. I will submit a pull request very soon.

c-dilks, 2025-04-11:
Seems like work on #558 is still ongoing; this issue will auto-close when it's merged.

closed 2025-04-14 by baltzell