JeffersonLab/coatjava #513: Missing zero suppression

whit2333, 2025-03-27:
https://github.com/JeffersonLab/coatjava/blob/3b01ae6fa746246c375c9ef791e164091a531bd5/common-tools/clas-detector/src/main/java/org/jlab/detector/decode/CodaEventDecoder.java#L815

Should this variable actually be called "firstSample" rather than "firstchannel"? If so them I think the zero suppression information is all here.

baltzell, 2025-04-04:
I think @ftouchte addressed this?

ftouchte, 2025-04-05:
Yes, I did. 

The first channel refers to the time of the first sample (expressed in term of bin number) according this picture.

[image]

Here is a little study.

**ZS=1 && Zstyp=1** run 21183 Here, wftime varies between 0 to 49

[image]

**ZS=1 && Zstyp=0** run 21175 Here wftime is always 0

[image]

**ZS=0** run 21123 Here wftime is always 0

[image]

whit2333, 2025-04-06:
OK. I was pointing out the variable name is confusing and probably misleading at best. Closing...

closed 2025-04-06 by whit2333