JeffersonLab/coatjava #580: AHDC helix fitter printing error

N-Plx, 2025-04-20:
The AHDC service prints an error whenever there are not enough point for the helix track fitter:

[https://github.com/JeffersonLab/coatjava/blob/f772c46c36298c85246c30528f7bda11da2b[…]lert/src/main/java/org/jlab/rec/ahdc/HelixFit/HelixFitJava.java](https://github.com/JeffersonLab/coatjava/blob/f772c46c36298c85246c30528f7bda11da2b06f0/reconstruction/alert/src/main/java/org/jlab/rec/ahdc/HelixFit/HelixFitJava.java#L242)

This has been a problem cooking data, creating very heavy log files that fill up the log directory for some runs where this was printed every event. 

A temporary solution could be to simply remove printing that line.

But it also needs to be checked why there are no good event, and how to deal with them. And if it is worth cooking with the AHDC service, if other information from the track finding or KF can be still looked at.

ftouchte, 2025-04-21:
It is true that we have a lot of bad events. Preventing them will same a lot of computed time and resources. I need to look deeply in the code. @RDupre 

[image]

N-Plx, 2025-04-21:
@efuchey

efuchey, 2025-04-21:
if there are too many hits, The conventional track finder seems to be used: 
https://github.com/efuchey/coatjava/blob/development/reconstruction/alert/src/main/java/org/jlab/service/ahdc/AHDCEngine.java#L177
That means there might not be a problem on AI track finder, but the fact we use the conventional track finder means we need to give it a bit of attention

PR #586 (2025-04-21): use logger
probably closes #580
closed 2025-04-21 by baltzell